paulo-raca/YetAnotherArduinoPcIntLibrary #3: Support for AVR ATMega 328PB

davija, 2020-12-16:
I'm working with an ATMega328PB which is very similar to the ATMega328P but I am getting an unsupported error.

paulo-raca, 2020-12-16:
Hello,

You'll need to update `PinChangeInterruptBoards.h`.

I believe it should be as simples as add `|| defined(__AVR_ATmega328PB__)` at the end of the first `#if`, can you try it?

davija, 2020-12-16:
I tried that and it ended up clobbering the entire port (I only needed interrupts on 3 pins).

I initially just setup an ISR on 3 pins (pins 16, 17 & 18) but I wasn't able to fully test  as it caused i2c to fail (also on the same port but different pins).

I didn't take much time to investigate, but I manually setup the port and pin masks and things worked fine, so I suspect there may be something a little off with this lib.  I'll take a look and see if I can do some more investigation later.

paulo-raca, 2020-12-16:
What do you mean with clobbering? 
The library should be able to use just a handful of pins without compromising the other pins in the port.

Can you check the implementation of digitalPinToPCICR,  digitalPinToPCMSK, digitalPinToPCICRbit and digitalPinToPCMSKbit? 

Those should be provided by arduíno's board headers